nasa-jpl/open-source-rover #475: No part numbers for Molex 6-pin connector

robrussell, 2024-04-14:
The [wiring instructions](https://github.com/nasa-jpl/open-source-rover/blob/master/electrical/wiring/README.md) require six 6-pin connectors made by Molex. The [parts list](https://github.com/nasa-jpl/open-source-rover/blob/master/parts_list/README.md#parts-for-electrical-assembly) currently lists only the mating part for the PCB - [Molex 1726750613](https://www.digikey.com/en/products/detail/molex/1726750613/5116933).

This came out of [a thread on Slack](https://opensourcerover.slack.com/archives/C01D07KNKJM/p1712119145360749). I'm working on a PR to update the relevant docs but I wanted to collect the info in one spot first.

### Parts

So I ordered some [Molex 0469920610 housings](https://www.digikey.com/en/products/detail/molex/0469920610/5116009) and pins [0039000077](https://www.digikey.com/en/products/detail/molex/0039000077/1643440) from Digikey. They seem to fit but I've only made one cable with them so far. I used the same [crimp tool](https://www.amazon.com/gp/product/B00OMM4YUY) as I did for the DuPont and JST-XH connectors. It wasn't perfect but seems workable. The housings are black so I can't mark them following the instructions.

### Assembled alternative

There is also a pigtail cable assembly in various lengths. I ordered [Molex 2153261063](https://www.digikey.com/en/products/detail/molex/2153261063/13883006) which is 60cm long and uses 16 AWG wire. The assembly could be more convenient but: 1. the assemblies seem relatively expensive, 2. they're less available, and 3. the wire gauge seems like it might be too large for the DuPont pins.

### Connector part names

To be explicit (so I don't have to look it up again) this connector housing is described by Molex as a Mini-Fit Jr. Receptacle Housing, Dual Row, Nylon 6/6, UL 94V-2, 6 Circuits, Glow-Wire Capable, Black from the series Mini-Fit Jr 46992. And the pins are Mini-Fit Jr. Overall Tin female crimp terminals from the series Mini-Fit Jr 5556.

These are also sometimes called PCIe power connectors because the same form factor is used in PC power supplies for supplemental power to video cards. These are currently the only Molex brand connectors in the OSR project but Molex (just like JST) make a whole lot of connectors, including some that are used for PCIe and other PC power connections.

Achllle, 2024-11-21:
@robrussell can we realistically still expect a PR from you to solve this one?

robrussell, 2024-12-11:
Hi @Achllle ! I added links to the Digikey parts to the docs in #476 . I didn't reference this issue though so I'm not sure if there was more to do here.

Achllle, 2024-12-11:
Got it, sounds like this can be closed then? If not, please reopen

closed 2024-12-11 by Achllle