nasa-jpl/open-source-rover #480: Update docs to use new bearing

abust005, 2024-07-01:
Bearing used in rocker-bogie ( [1601-0014-0008](https://www.gobilda.com/1601-series-flanged-ball-bearing-8mm-id-x-14mm-od-4mm-thickness-2-pack/) )has been discontinued and replaced with [1611-0514-0008](https://www.gobilda.com/1611-series-flanged-ball-bearing-8mm-id-x-14mm-od-5mm-thickness-2-pack/).

This may require testing and validation, as the replacement is 1mm thicker.

Achllle, 2024-07-08:
Tried replacing within OnShape. The spacers are a fraction of a mm too thick now, but I believe that won't cause much of an issue. Will submit a PR for the change, let me know if you think this will create issues.
[image]

abust005, 2024-07-08:
Looks good to me, if anything you might be right with what you said in the Slack, this could actually be positive.

Achllle, 2024-07-08:
I just noticed that the [parts list](https://github.com/nasa-jpl/open-source-rover/tree/master/parts_list#parts-for-rocker-bogie-assembly) is already pointing to the newer version. It seems like only the OnShape model was outdated.

closed 2024-07-08 by Achllle