nasa-jpl/open-source-rover #482: Section 4.5 of PCB assembly instructions is a little confusing

ldang, 2024-09-21:
While we were soldering the brainboard, section 4.5 was a bit confusing due to how the [instructions](https://github.com/nasa-jpl/open-source-rover/blob/master/electrical/pcb/README.md) were worded.

Originally, with the many reference to E-STOP and LED indicators, our eyes kept being drawn to T13, T14, T15, and we wondered what component was supposed to be installed there.

After much discussion among the group and rereading of the instructions, we believe section 4.5 was about installing the transistors (Q2-Q6) and the diodes (D8 and D9).   And rereading the earlier instructions, we can see that other T* references were called test points.

So reasons we are confused:

In previous instructions, the bullets indicated the components (using their reference part number) to be installed, without a lot of verbiage.   In section 4.5, there are a bunch of words about E-STOP and then the actual transistor part numbers.  The diodes aren't even in the bullets but in a separate paragraph.

Additionally, since we are soldering the PCB over several weeks, we tend to focus on the section in front of us, and may not remember that you have been consistently using T* designators for test points.  So we may need a reminder that T13, T14, T15 are test points for the E_STOP circuit.  Additionally, it would be nice to have a test procedure for that in this section, even if we may not be able to test the circuit.

Secret-Asian-Man, 2024-09-24:
Submitted pull request https://github.com/nasa-jpl/open-source-rover/pull/483 for this issue.

Secret-Asian-Man, 2024-10-18:
This issue this was solved by #483 
@Achllle please close thanks.

closed 2024-10-18 by Achllle