nasa-jpl/open-source-rover #484: Hard to notice OnShape model + Slack community link on REAMDE

ldang, 2024-09-27:
This is something I've noticed when building the rover and also from talkign to other people building the rover.

For some reason, it's difficult to notice that there is an OnShape 3D CAD model we can use.  And also, it's unclear how to join the Slack.

I think it might be an issue of the information not being located in the right area of the document to be relevant to the user.  Most of the time, we end up skimming the document for the information of interest, and then read it more carefully from there.  Being reminded of the OnShape model would be most useful around the Rover Build Map section.

For the Slack issue, I think people who are not familiar with Slack will not understand how to join Slack, and also, it's unexpected that a giant Slack logo is an active link.

Generally, it's better to be repetitive for important pieces of information, so there are multiple places for people to get the information, even if they skimmed over the place where it's explained in detail.

Achllle, 2024-09-27:
The OnShape model is both mentioned on the main page in its own section, and on the mechanical README page:

> Along with the pictures in each folder's README document, open the [OnShape model](https://cad.onshape.com/documents/e4f00b1a3d2edb1a84bbba1c/w/8ab8f394324bcc586236ef5d/e/9191e5ad2a70f387b419bc55?renderMode=0&uiState=645ede92f3a1a9205158b296) so you can see what goes where and spin it around. Note that the OnShape model does not have all screws included to keep the model speedy and light.

in the mechanical/body/README page:

> pay close attention to the placement of the beams using the image below and the online OnShape model.

twice in the rocker_bogie README:

> > **Tip**: Use the OnShape model and navigate to the tab 'rocker-bogie' in the bottom to isolate this assembly. Then right click on various parts to temporarily hide them so you can see how the assembly works in 3D.

> > **NOTE**: the OnShape model may represent the orientation of the slits incorrectly. In order to make the model nimble, the same rocker-bogie assembly is used on both sides of the rover which has the side-effect that the two assemblies aren't mirrorred like they are supposed to be. Always stop to think about what you're building to avoid having to take things apart later.

I agree that the Slack link could be in text as well. PR welcome. Similar to the above, Slack is suggested 4 other times throughout the instructions in the parts_list, examples, rocker_bogie READMEs.

Achllle, 2024-11-21:
If you have specific suggestions for how to improve this, please add them here or better create a pull request. Closing this until then so we don't have a bunch of stale issues.

closed 2024-11-21 by Achllle