nasa-jpl/open-source-rover #490: Clarify in electrical/pcb/README.md that the INA260 needs to be configured to not use default I2C address

apollokit, 2024-10-16:
Need to add jumpers onto the INA260 to change its I2C address. Update the instructions to reflect this

Achllle, 2024-11-12:
@apollokit is this realistic for you to take on in the next few weeks?

apollokit, 2025-10-08:
Yep, per Step 5 under [subsection 3.5 in the pcb assembly instructions](https://github.com/nasa-jpl/open-source-rover/blob/master/electrical/pcb/README.md#35-install-the-ina-260) the address to use is is 0x45

Honestly I'm not really understanding anymore why I opened this issue - the instructions look like they're up to date already? I soldered across the pads to jump it, didn't use "jumpers" per se, but I think the instructions are good? Admittedly it's been a while...