nasa-jpl/open-source-rover #493: Add lego-style instructions

ldang, 2024-11-22:
The Beyond Robotics FTC team made a [Google doc](https://docs.google.com/document/d/1aSEBnRjCp-DHMCkOAVH4jRNDluBRjEW73nd1NysC_DA/edit?usp=sharing) with lego-style instructions.

I think we can incorporate this into the our current repo by using pandoc to convert it to Markdown.  I'm not sure where it should fit in the repo structure and how to get it into read-the-docs.

igillespie, 2024-11-28:
As someone who just ordered all of the parts to build this at home with two kids. These instructions make this project seem way more doable. Thank you!

javanniissaih26, 2026-08-18:
To integrate the LEGO-style instructions effectively, we could create a subdirectory within `docs/` to keep the original Google doc organized. After using pandoc to convert it to Markdown, we can name the file something like `lego_instructions.md`. To ensure it gets properly linked in Read the Docs, we should add an entry for it in the `index.rst` file under a new section, perhaps titled "Assembly Instructions." Is there any specific update format for `index.rst` that we should follow to maintain consistency with existing documentation?

Achllle, 2026-08-20:
That doesn't solve the 'second set of documentation' problem. Closing.

closed 2026-08-20 by Achllle