nasa-jpl/open-source-rover #499: switch cutout for back body plate doesn't fit switch

Achllle, 2024-12-17:
[image]

I created a temporary 3D print insert [here](https://cad.onshape.com/documents/2ed88c9fc0b47bffbc3d1d97/w/ced9ad329dea5ab1f8186685/e/a090dcfebe2e4591f6a02445?renderMode=0&uiState=6761a386aa2aef4bcc9706d0) but the original files for laser cutting should be updated.

tayvionlucciano26, 2026-08-29:
The switch cutout on the back body plate is clearly undersized relative to the dimensions in the parts list, which is what forced the temporary Onshape insert. We should take the exact outline from that linked model and apply it directly to the laser cutting source files rather than treating the insert as a permanent bandaid. If those files are generated from a separate CAD or DXF that hasn't been synced to the current switch spec, that would explain the mismatch going forward.

takeishataralyn26, 2026-09-09:
I will verify the Onshape sketch geometry against the DXF source file dimensions to confirm the exact offset required for the laser kerf adjustment.

darnellaakaisha26, 2026-10-28:
The current laser cut files likely assume a standard rectangular tolerance, but the switch housing appears to require additional clearance for the side-locking tabs visible in the image. Could you confirm if the current DXF files include an offset for the kerf that might be shrinking the effective cutout dimension below the switch's outer shell measurements?