nasa-jpl/open-source-rover #516: XBox Dongle Discontinued/Bluetooth Controller Upgrade

idgaron, 2025-05-29:
The Github mentions using an xbox controller with the xbox usb dongle to connect bluetooth over usb instead of the internal bluetooth; however, this dongle is discontinued by Microsoft. A better alternative would be to use a [bluetooth module](https://www.adafruit.com/product/1327). This will allow users to use any bluetooth controller by switching the bluetooth interface of the pi from the UART lines to the USB lines. I have successfully used this on my rover build. Thoughts?

Achllle, 2025-05-29:
Even if Microsoft discontinues their lineup, there's always a huge aftermarket. I believe they all use bluetooth anyways. [8bitdo](https://shop.8bitdo.com/products/8bitdo-wireless-usb-adapter-2-for-switch-windows-mac-raspberry-pi-compatible-with-xbox-series-x-s-controller-xbox-one-bluetooth-controller-switch-pro-and-ps5-controller?gQT=1) makes good ones as well. PRs welcome!

tayvionlucciano26, 2026-08-18:
One consideration when switching from the Xbox dongle to the Bluetooth module is ensuring that the `bluetooth.py` handlers accommodate the change in the underlying Bluetooth interface. You might need to modify the connection logic in `connect_controller()` to properly initialize and manage the new module. Also, are there any specific configurations or libraries you've used to facilitate this transition on the Raspberry Pi? It would be helpful to document those for users who might face similar challenges.

laveahhaneef26, 2026-08-18:
When switching to the Bluetooth module from the Xbox dongle, have you tested how the `connect_controller()` function handles the initialization process? It might require adjustments to check for the module's specific connection parameters, especially if you're intending to support multiple Bluetooth controllers. Additionally, ensuring that the authentication process aligns with the new hardware is crucial—perhaps adding a fallback mechanism in case the connection times out could enhance reliability.